MicrosoftDocs/windows-driver-docs-ddi #160: ULONG are UINT16

mseld2, 2018-07-27:
When I looked in memory it looks like each field in the struct is a UINT16, not ULONG as the documentation states. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0db27d7d-9ed3-cf8b-9518-6d6b22d1ab16
* Version Independent ID: a7a9597b-5bd8-dcbf-7bc8-13405a69584f
* Content: [_USB_PORT_PROPERTIES](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/usbioctl/ns-usbioctl-_usb_port_properties)
* Content Source: [wdk-ddi-src/content/usbioctl/ns-usbioctl-_usb_port_properties.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/usbioctl/ns-usbioctl-_usb_port_properties.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

ajbarb, 2018-07-30:
The debugger output below shows that the ul is ULONG:

5: kd> dt USB_PORT_PROPERTIES
UsbHub3!USB_PORT_PROPERTIES
   **+0x000 ul               : Uint4B**
   +0x000 PortIsUserConnectable : Pos 0, 1 Bit
   +0x000 PortIsDebugCapable : Pos 1, 1 Bit
   +0x000 PortHasMultipleCompanions : Pos 2, 1 Bit
   +0x000 PortConnectorIsTypeC : Pos 3, 1 Bit
   +0x000 ReservedMBZ      : Pos 4, 28 Bits

Can you clarify with debugger output as to where you see UINT16.

mseld2, 2018-08-02:
I'm using pinvoke in C#. I initially Marshalled my structure according to the documentation, but it wasn't working, so I looked in the memory. This is what I saw:

0x0000023B3C2BAEA0  **02 00 00 00 98 00 00 00 01 00 00 00 00 00 0f 00** 55 00 53 00 42 00 23 00 52 00 4f 00 4f 00 54 00 5f 00 48 00 55 00 42 00 33 00 30 00 23 00 34 00 26 00 61 00  ....˜...........U.S.B.#.R.O.O.T._.H.U.B.3.0.#.4.&.a.
0x0000023B3C2BAED4  66 00 66 00 62 00 65 00 39 00 37 00 26 00 30 00 26 00 30 00 23 00 7b 00 66 00 31 00 38 00 61 00 30 00 65 00 38 00 38 00 2d 00 63 00 33 00 30 00 63 00 2d 00  f.f.b.e.9.7.&.0.&.0.#.{.f.1.8.a.0.e.8.8.-.c.3.0.c.-.
0x0000023B3C2BAF08  31 00 31 00 64 00 30 00 2d 00 38 00 38 00 31 00 35 00 2d 00 30 00 30 00 61 00 30 00 63 00 39 00 30 00 36 00 62 00 65 00 64 00 38 00 7d 00 00 00 00 00 00 00  1.1.d.0.-.8.8.1.5.-.0.0.a.0.c.9.0.6.b.e.d.8.}.......

As you can see, the offset to the CompanionHubSymbolicLinkName is 16 bytes (see bolded bytes above). If I were to Marshal my structure according to the documentation, the offset would be 20 bytes, which would put me 4 bytes into the CompanionHubSymbolicLinkName . Based on what I saw in memory, I decided to Marshal the structure like this:

       [StructLayout(LayoutKind.Sequential)]
        public struct USB_PORT_PROPERTIES
        {
            public ushort ul; // [8:9] 01 00 
            public ushort Properties; //  [10:11] 00 00
        }

        [StructLayout(LayoutKind.Sequential, CharSet = CharSet.Auto)]
        public struct USB_PORT_CONNECTOR_PROPERTIES
        {
            public uint ConnectionIndex; // [0:3] 02 00 00 00
            public uint ActualLength; // [4:7] 98 00 00 00
            public USB_PORT_PROPERTIES Properties;  // [8:11]
            public ushort CompanionIndex; // [12:13] 00 00
            public ushort CompanionPortNumber;// [14:15] 0f 00
            [MarshalAs(UnmanagedType.ByValTStr, SizeConst = 255)]
            public string CompanionHubSymbolicLinkName; // [16:end] 
        }

This works fine. I get appropriate values for all fields in the structure, and I get the CompanionHubSymbolicLinkName as expected.

mseld2, 2018-08-02:
Oh, I'm so sorry! I looked at the documentation of the structure again because something didn't seem right with my results compared to your output. I missed the **UNION** part of it. In that case, it's 4 bytes, as you said

I apologize for the confusion. Duh! Any way, I changed my structure to be Marshalled as shown below and it works just peachy.

[StructLayout(LayoutKind.Sequential, CharSet = CharSet.Auto)]
    public struct USB_PORT_CONNECTOR_PROPERTIES
    {
        public uint ConnectionIndex; 
        public uint ActualLength;
        public uint Properties; 
        public ushort CompanionIndex;
        public ushort CompanionPortNumber;
        [MarshalAs(UnmanagedType.ByValTStr, SizeConst = 255)]
        public string CompanionHubSymbolicLinkName; 
    }

Again, I'm very sorry you spent time on something that was due to my bad speed-reading skills.

EliotSeattle, 2018-08-02:
Thanks, @mseld2 
Glad the docs are right and that they all lived happily ever after. At least in this respect. Cheers!

closed 2018-08-02 by EliotSeattle