MicrosoftDocs/windows-driver-docs-ddi #162: NdisMQueueDpcEx return value clarification

I-gor-C, 2018-08-01:
[NdisMQueueDpcEx documentation](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ndis/nf-ndis-ndismqueuedpcex) states:

> If the driver requested a DPC for a CPU, and NDIS indicates that it did not schedule that DPC, the DPC was not scheduled because a DPC was already scheduled for that CPU.

Why would a CPU with already scheduled DPC prevent another DPC from being scheduled to it?
Could someone clarify the meaining here?

duncanmacmichael, 2018-08-01:
Hi @I-gor-C , good question. I'll ask the NDIS team to see what the original intention behind that sentence is, and update it if need be.

duncanmacmichael, 2018-08-02:
After some discussion, the NDIS team agrees that the wording here is a bit vague. Saying "**a** DPC was already scheduled" makes it sound like NDIS couldn't schedule the DPC due to some other random DPC being there already. 

The true meaning is that NDIS has one DPC object for this interrupt/processor pair, so if _that_ DPC is already scheduled then NDIS won't schedule it. Does this wording make more sense?

"If the driver requested a DPC for a CPU, and NDIS indicates that it did not schedule that DPC, then the DPC was not scheduled because the DPC that maps to this interrupt/processor pair was already scheduled for that CPU."

I-gor-C, 2018-08-03:
Makes perfect sense, thanks!

closed 2018-08-03 by I-gor-C

duncanmacmichael, 2018-08-03:
Great! You're very welcome. I'll go ahead and update the page, then. Thanks for pointing this out!