MicrosoftDocs/windows-driver-docs-ddi #163: Clarification needed regarding validity of signatures

cyren-oschneider, 2018-08-02:
`CertificateThumbprint`, if non-`NULL` seems to contain the hash on the signature inside the certificate used for signing an image.

Now, I could not find any guidance (or even remarks) on whether it would be a reasonable approach to keep a list of those thumbprints as well (signed and to be verified, just like the image hashes would have to be) and consider them for classifying the image at hand?

So my idea here is as follows:

1. check the image hash against my list of hashes to find [the appropriate classification](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/ne-ntddk-_bdcb_classification)
1. if the image hash is not on my list, see if `CertificateThumbprint` is not `NULL` and if it isn't, compare against my list of **certificate thumbprints** and derive a classification from that.

This is not unlike how the chain of trust works in SecureBoot, if I don't completely misunderstand _that_. It also relies on items to be signed and likely verifies validity of the signatures. What's good enough for SecureBoot should be good enough as a fall back in ELAM for otherwise unknown images (e.g. because a system update was applied, but no update to the ELAM image hash list).

Of course for such a method to work as step 2 two things would have to be taken into account:

1. I must be sure that the signature is valid
1. I need a list of such thumbprints

While it's possible to get these thumbprints ourselves, it would possibly be better for Microsoft to provide such a list - as far as shipped drivers (installation media and Windows Update) are concerned to us vendors. Of course if only the combination of thumbprint, issuer and publisher is secure enough, that would also work. The general idea should not be affected by it.

In fact I'd go so far as to suggest that Microsoft store such a list of hashes, signed by a Microsoft key, inside the ELAM hive's root and provide the MVI vendors with the public key in order to be able to verify the signed list of thumbprints.

However, back to the point about the validity of the signature. If `ImageFlags` has bit 1 cleared, does that mean the signature has been verified already and our ELAM can rely on that information?

Thanks and with best regards,

Oliver

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4080585d-6e46-b6c7-115d-dfa97477eb33
* Version Independent ID: 181e5da7-4032-e667-31af-14460bf72d4a
* Content: [_BDCB_IMAGE_INFORMATION](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/ns-ntddk-_bdcb_image_information)
* Content Source: [wdk-ddi-src/content/ntddk/ns-ntddk-_bdcb_image_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/ns-ntddk-_bdcb_image_information.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-02:
Thanks, @cyren-oschneider for the thoughtful feedback
@PageWriter-MSFT can you please triage this request and determine what might affect the docs, what should be forwarded to the dev team, and what might best be answered in a forum?

tedhudek, 2019-05-31:
Looks like the opener of this feedback has deleted his/her GitHub account.

Regarding ImageFlags, yes if bit 1 is clear then it means Code Integrity found a valid signature for the file. 

However, that should be taken with a grain of salt. For example, if Secure Boot is not enabled then the Boot Manager and Winload could have been tampered with to report everything is valid. The chain of trust needs to be validated.

I did forward your suggestion about providing a list of hashes to the ELAM team.

closed 2019-05-31 by tedhudek