MicrosoftDocs/windows-driver-docs-ddi #164: BDCB_IMAGEFLAGS_FAILED_CODE_INTEGRITY should be mentioned

cyren-oschneider, 2018-08-02:
Hi,

I looked at `ntddk.h` in the vicinity of `_BDCB_IMAGE_INFORMATION` and found that there is in fact a symbolic name for bit 1 of the `_BDCB_IMAGE_INFORMATION::ImageFlags` field. It would be nice to mention that in the documentation.

**It would also be nice to assign a symbolic name to bit 0.** (probably not a task for the doc writer, though)

Excerpt from `ntddk.h`:

```
//
// BDCB_IMAGE_INFORMATION ImageFlags.
//

#define BDCB_IMAGEFLAGS_FAILED_CODE_INTEGRITY   (1UL << 0)
```
Assuming this flag didn't exist in earlier incarnations of the Windows 10 WDK (I don't have earlier WDKs on this machine) it's easy for us driver writers to guard against such a case, say, by doing this _in my own code_ (after including `ntddk.h`):

```
#ifndef BDCB_IMAGEFLAGS_FAILED_CODE_INTEGRITY
#define BDCB_IMAGEFLAGS_FAILED_CODE_INTEGRITY   (1UL << 0)
#endif
```

Thanks and with best regards,

Oliver

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4080585d-6e46-b6c7-115d-dfa97477eb33
* Version Independent ID: 181e5da7-4032-e667-31af-14460bf72d4a
* Content: [_BDCB_IMAGE_INFORMATION](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/ns-ntddk-_bdcb_image_information)
* Content Source: [wdk-ddi-src/content/ntddk/ns-ntddk-_bdcb_image_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/ns-ntddk-_bdcb_image_information.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-02:
Thanks, @cyren-oschneider 
@PageWriter-MSFT can you please look into this and also passing the one comment to the dev team?

tedhudek, 2019-05-20:
Added the symbolic name to the doc page.  The update will go live later today.

closed 2019-05-20 by tedhudek