MicrosoftDocs/windows-driver-docs-ddi #166: Return code is wrong

hsebs, 2018-08-03:
There are two STATUS_BUFFER_OVERFLOW codes in Return Value section.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5966a2f8-9648-ff27-5f70-3a9ef9e2eb59
* Version Independent ID: 7429b5de-0a58-a955-ecc5-87c897528e73
* Content: [WdfRegistryQueryValue function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdfregistry/nf-wdfregistry-wdfregistryqueryvalue)
* Content Source: [wdk-ddi-src/content/wdfregistry/nf-wdfregistry-wdfregistryqueryvalue.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfregistry/nf-wdfregistry-wdfregistryqueryvalue.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

hsebs, 2018-08-03:
The second one might be STATUS_BUFFER_TOO_SMALL.

EliotSeattle, 2018-08-03:
Thanks, @hsebs 
@tedhudek can you correct the return value, please?

tedhudek, 2018-09-12:
Hi @hsebs sorry for the delay on this one.  After a bit of investigation, I am fairly certain that the listed return values are correct.  Please don't hesitate to re-open if you have actual test evidence to the contrary.  Thanks for reaching out!

closed 2018-09-12 by tedhudek