MicrosoftDocs/windows-driver-docs-ddi #179: Syntax Error

agamagtdM10102148, 2018-08-06:
Actual code in wdm.h (SDK, WDK 15063)

__forceinline
UCHAR
READ_PORT_UCHAR (
    _In_ _Notliteral_ PUCHAR Port
    )

{
    UCHAR Result;

    _ReadWriteBarrier();
    Result = __inbyte((USHORT)((ULONG_PTR)Port));
    _ReadWriteBarrier();
    return Result;
}

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7d0b8088-e158-db22-8c5a-7de06a38c2f9
* Version Independent ID: a56d09c7-52b5-72f9-568e-1483f925bea4
* Content: [READ_PORT_UCHAR function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ioaccess/nf-ioaccess-read_port_uchar#feedback)
* Content Source: [wdk-ddi-src/content/ioaccess/nf-ioaccess-read_port_uchar.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ioaccess/nf-ioaccess-read_port_uchar.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

agamagtdM10102148, 2018-08-06:
Sorry, 
identical function name in different header file

EliotSeattle, 2018-08-06:
Thanks, @agamagtdM10102148 
If I read your second comment correctly, you are OK? Go ahead and reopen this issue if you still need clarification.

closed 2018-08-06 by EliotSeattle