MicrosoftDocs/windows-driver-docs-ddi #180: ReturnLength usage is incorrect

randomascii, 2018-08-06:
This sentence is incorrect: "If the entire region of pages does not have a matching set of attributes, then the ReturnLength parameter value can be used to compute the address and length of the region of pages that was not scanned."

The ReturnLength parameter refers to how many bytes are copied to the MemoryInformation pointer. It does not refer to the size of memory regions. Or, the documentation of ReturnLength in the list of parameters is wrong. One of these mentions of ReturnLength is incorrect.

Also, I'm calling this function and getting a return value of 0xC0000005, which is not a documented return code. It would be nice to document that.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7a307db7-3d17-d025-8c04-5dd80bddf8cc
* Version Independent ID: c7c80dd0-04a2-5c0b-a486-14d123f8f74f
* Content: [NtQueryVirtualMemory function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-ntqueryvirtualmemory)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-ntqueryvirtualmemory.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-ntqueryvirtualmemory.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-08:
Thanks for catching this, @randomascii 
I've reached out to the dev team to disambiguate the descriptions and to check on the return value. I'll update the topic as soon as I get some info.

randomascii, 2018-08-08:
I got my code working so I can say with a fair degree of certainty that the documentation should say something like this:

If the entire region of pages does not have a matching set of attributes then the sub-regions within one address reservation will be returned individually. They will all have the same info.AllocationBase value, their individual sizes will be in info.RegionSize, and the total size of the address reservation can be obtained by summing all of the info.RegionSize values associated with a particular info.AllocationBase value.

Sample code that implements this can be found in https://bugs.chromium.org/p/chromium/issues/detail?id=870054#c16 in VirtualScan.zip.

lorihollasch, 2019-06-08:
@randomascii , thank you for reporting this, and for your subsequent proof-of-concept work to verify your suggested update. I have added your update to the refpage, and added the error code description you thought would be useful. SO SORRY that it took us so long to complete and close this issue!!

closed 2019-06-08 by lorihollasch