MicrosoftDocs/windows-driver-docs-ddi #181: TBD for parameters

hsebs, 2018-08-07:
The description of DriveLayout in Parameters is TBD.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1c04966c-6cd6-8d51-2795-7ac229c674a5
* Version Independent ID: ec532a3d-f5e9-5995-7eed-f5217bf21014
* Content: [IoReadPartitionTableEx function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/nf-ntddk-ioreadpartitiontableex)
* Content Source: [wdk-ddi-src/content/ntddk/nf-ntddk-ioreadpartitiontableex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/nf-ntddk-ioreadpartitiontableex.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-07:
Thanks for pointing this out, @hsebs 
@PageWriter-MSFT can you please create the description for the DriveLayout parameter here?

tedhudek, 2019-05-20:
Fixed

closed 2019-05-20 by tedhudek