MicrosoftDocs/windows-driver-docs-ddi #188: The description is missing.

hsebs, 2018-08-12:
The description of PartitionType is missing in Members section.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 96478538-5b41-add4-a553-6e289b730642
* Version Independent ID: 0ab52842-d992-b503-e7f8-2f875a05274e
* Content: [_PARTITION_INFORMATION](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntdddisk/ns-ntdddisk-_partition_information)
* Content Source: [wdk-ddi-src/content/ntdddisk/ns-ntdddisk-_partition_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntdddisk/ns-ntdddisk-_partition_information.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-13:
Thanks, @hsebs 
I'll need to take a look at this and will update it as I have the time. Cheers!

EliotSeattle, 2019-01-09:
This was completed on 12/19, but I forgot to close the Issue.
Thanks for submitting this, @hsebs!

closed 2019-01-09 by EliotSeattle