MicrosoftDocs/windows-driver-docs-ddi #189: An information is missing.

hsebs, 2018-08-12:
Line 4523~4533 of fltKernel.h in WDK 10.0.17134.0

NTSTATUS
FLTAPI
FltSendMessage (
    _In_ PFLT_FILTER Filter,
    _In_ PFLT_PORT *ClientPort,
    _In_reads_bytes_(SenderBufferLength) PVOID SenderBuffer,
    _In_ ULONG SenderBufferLength,
    _Out_writes_bytes_opt_(*ReplyLength) PVOID ReplyBuffer,
    _Inout_opt_ PULONG ReplyLength,
    _In_opt_ PLARGE_INTEGER Timeout
    );

But it is not description that ReplyLength is optianal parameter in Parameters Section.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3f5636eb-7229-0c98-d614-48e26e1185c9
* Version Independent ID: b20a4670-f96f-81b1-ebbb-0cdc06462def
* Content: [FltSendMessage function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nf-fltkernel-fltsendmessage)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltsendmessage.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltsendmessage.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-13:
Thanks, @hsebs 
I'll do some research and get this fixed as soon as I can.

lorihollasch, 2019-05-14:
@hsebs, thanks for reporting this and sorry it took us so long to address this! I've updated the FltSendMessage page with your suggestions, and made some additional clarifying changes.

closed 2019-05-14 by lorihollasch