MicrosoftDocs/windows-driver-docs-ddi #191: Characteristics not defined

BrianCatlin, 2018-08-15:
The description for wHubCharacteristics is "The hub characteristics". Really? Could you be any more lazy and uninformative? How about a list of the bit fields? They have to be in one of the Microsoft header files somewhere because you have a hub driver that uses this field

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b1e953a3-9c63-7511-02a3-ba8bebb78628
* Version Independent ID: 3a81f94c-c2d5-6def-587a-0796f8a8e547
* Content: [_USB_30_HUB_DESCRIPTOR](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/usbspec/ns-usbspec-_usb_30_hub_descriptor)
* Content Source: [wdk-ddi-src/content/usbspec/ns-usbspec-_usb_30_hub_descriptor.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/usbspec/ns-usbspec-_usb_30_hub_descriptor.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-15:
Thanks, @BrianCatlin 
We usual don't strive for that. Sometimes it happens in spite of our good intentions. 
@PageWriter-MSFT can you please see if you can dig up some bit fields or other more descriptive phrases to add to that member?

PageWriter-MSFT, 2018-09-13:
Hi Brian,

I have added the section in the USB specification that describes the hub descriptor in detail. In general, for such spec structures, we mention the sections from the specification to avoid maintenance nightmares.  I will admit we haven't been consistent with that practice. We'll try to do better in the future.

Page Writer

closed 2018-09-13 by PageWriter-MSFT