MicrosoftDocs/windows-driver-docs-ddi #193: Enumerator mismatch

beemavishnu, 2018-08-16:
Hello,

In above , enumerators start from 0.

Where as in WDK10 wditypes.hpp enumerators start from 1 as shown below.

typedef enum _WDI_SCAN_TRIGGER
{
    WDI_SCAN_TRIGGER_MANUAL = 1,
    WDI_SCAN_TRIGGER_BACKGROUND = 2,
    WDI_SCAN_TRIGGER_ROAM = 3,
    WDI_SCAN_TRIGGER_CONNECT = 4,
    WDI_SCAN_TRIGGER_ANQP_QUERY = 5,
    WDI_SCAN_TRIGGER_FAST_ROAM = 6
} WDI_SCAN_TRIGGER;

Please update the content in this link.

Thanks & Regards
Vishnu Beema

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7bd0149d-80a2-5b0b-8731-55a33d2afaac
* Version Independent ID: da88d095-d614-c1bc-224d-868069047c8c
* Content: [_WDI_SCAN_TRIGGER](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wditypes/ne-wditypes-_wdi_scan_trigger)
* Content Source: [wdk-ddi-src/content/wditypes/ne-wditypes-_wdi_scan_trigger.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wditypes/ne-wditypes-_wdi_scan_trigger.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-16:
Thanks, @beemavishnu for catching this. 
@duncanmacmichael can you please look into this and make sure our numbering is correct? What a difference an integer makes...

duncanmacmichael, 2018-08-16:
Hi @beemavishnu, you're right in that the enum values start at 1 in the header file, but the syntax on this topic doesn't reflect that. In all of our reference documentation (not just on this page), we currently do not support showing the integer values for enums. But, I'll open an internal work item to request that we add this in the future.

I'll close this issue for now. Thanks for your interest in the docs!

closed 2018-08-16 by duncanmacmichael