MicrosoftDocs/windows-driver-docs-ddi #194: Does not work correctly with the Unified Write Filter in some, potentially dangerous, cases.

alfaunits, 2018-08-16:
This API outputs a potentially dangerous result in combination with the Unified Write Filter.
If the root of the protected drive contains an excluded 5-character folder, attempting to query and parse the file name information on it (and only on it, not its subitems) results in -&gt;Volume.Length being greater than -&gt;Name.Length, and other fields being 0-length string, which is an obvious bug.
The dangerous part is that production code is unlikely to test such cases, and -&gt;Name.Length - -&gt;Volume.Length would result in -2, potentially caussing data overruns/underruns or memory access violations in drivers.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 936ffc60-fa65-57e0-7924-98fd42e9c39d
* Version Independent ID: 9fd65b9f-fa8a-6f83-7db7-03a08d84f022
* Content: [FltParseFileNameInformation function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nf-fltkernel-fltparsefilenameinformation)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltparsefilenameinformation.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltparsefilenameinformation.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

alfaunits, 2018-08-17:
Thanks. BTW, I have forgotten one important thing! The mini-filter needs to be above the UWF filter in altitude, so for testing, set it to 390000

Set up the UWF as following, it is the simplest way to configure it to reproduce the issue:
    uwfmgr.exe volume protect d:

    uwfmgr.exe file add-exclusion d:\12345

    uwfmgr.exe filter enable

Reboot of course ;)

Cheers.

alfaunits, 2018-10-23:
Any update on this?

EliotSeattle, 2018-10-23:
My bad. This slipped through the cracks. I've gona ahead and raised it with the dev on the IFS team.

EliotSeattle, 2018-10-24:
Hi @alfaunits.
Would the dev team be able to see a dump file from the occurance?

alfaunits, 2018-10-24:
We test for such cases, so there is no dump. Let me see if I can induce it
in a test env.

alfaunits, 2018-11-19:
Sorry for a late reply. I have been able to gather a dump manually.

Is there anywhere I can upload the dumps for you?

EliotSeattle, 2018-11-19:
Great! If you can put it in a zip file, you can send it to eliotgra @ microsoft . com

EliotSeattle, 2018-11-26:
Hi @alfaunits.
I had a dev look at your dump files, and they had this comment:

"They seem to have called FltGetFileInformtionUnsafe instead of FltGetFileInformation in their post create callback. Can you please help confirm this with them? FltGetFileInformationUnsafe should *not* be used on IO paths."

Is he correct?

Thanks,
@EliotSeattle

alfaunits, 2018-11-26:
It is possible, but only in caae FltGetFileNameInformation failed. I
cannot be sure whether it was called, because we don't track it. But,
yes, it is a possibility.

The logic is: if(!NT_SUCCESS(FltGetFileNameInformation))
FltGetFileNameInformationUnsafe

alfaunits, 2019-01-04:
Any update on this? Or was it considered closed, presuming FGFNIUnsafe must not be used?

EliotSeattle, 2019-01-07:
The latter is correct. Thanks!

closed 2019-01-07 by EliotSeattle

alfaunits, 2019-01-07:
That makes FltMgr's name management routines useless. Or the entire structure where the name cannot be queried.

What the hell is the FltMgr for then?

EliotSeattle, 2019-01-10:
Sorry that I'm not quite getting the right answers to you, @alfaunits. Can you please reach out to the DDI creators? you can find them at msfltdev @ microsoft . com. They've agreed to help you figure out why the safe name query failed in the first place (and seemingly consistently) and what to do to avoid that. Please reference this Issue with a link within your mail. 
All the best. 
Eliot

alfaunits, 2019-01-10:
Just did that.

On this topic, the Closure makes no sense. It is not the Unsafe variants of the FGFNI calls that fail, but it is FltParseFileNameInformation (a simple string processing API by all means) that  produces incorrect results.

EliotSeattle, 2019-01-10:
The devs on that list are by far the best people to help you out here. Where there are doc changes to make, I'm happy to review a PR or make the changes.

alfaunits, 2019-01-11:
Oh, Khm.. I did not even realize this is for Docs only ;)

I will try to make a repro case to see if this can happen any time with a skeleton driver, so that the docs would include the note, until a fix.

EliotSeattle, 2019-01-11:
Thanks, and let me know what you find, @alfaunits

alfaunits, 2019-02-02:
Just a note that we have tested a new build and confirmed that the code DID NOT call FltGetFileNameInformationUnsafe at all before the problematic FltParseFileNameInformation.
The parsing also does not fail when the file name format requested is OPENED, but only NORMALIZED (even though the strings are identical).

I am checking with the DDI team what is up.