MicrosoftDocs/windows-driver-docs-ddi #195: Not supported by root hubs

BrianCatlin, 2018-08-16:
The docs should explicitly mention that this IOCTL will return STATUS_UNSUCCESSFUL (What a horrible status to use! Why not STATUS_INVALID_DEVICE_REQUEST or, better yet, just return the topology info?) when sent to a root hub

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: cb6fdb5b-5b6d-4926-b55d-3d128a6c4c50
* Version Independent ID: 0f749787-7efb-7a6a-1776-b0b0facd8348
* Content: [IOCTL_INTERNAL_USB_GET_TOPOLOGY_ADDRESS](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/usbioctl/ni-usbioctl-ioctl_internal_usb_get_topology_address)
* Content Source: [wdk-ddi-src/content/usbioctl/ni-usbioctl-ioctl_internal_usb_get_topology_address.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/usbioctl/ni-usbioctl-ioctl_internal_usb_get_topology_address.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-16:
Thanks, @BrianCatlin 
@PageWriter-MSFT please clarify this topic.

BrianCatlin, 2018-08-17:
My testing shows that this works for non-USB3 root hubs. Please explain why it fails for USB3 root hubs, and how this bug can be worked around

closed 2019-05-15 by duncanmacmichael