MicrosoftDocs/windows-driver-docs-ddi #196: Wrong structure name

BrianCatlin, 2018-08-16:
Why is this page named for the structure tag (_HUB_DEVICE_CONFIG_INFO_V1) instead of the structure type (HUB_DEVICE_CONFIG_INFO)? None of the docs or examples ever refer to this structure using the tag

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1b88c77d-d9a8-9cfe-7cea-d2a502fb9248
* Version Independent ID: 7dbc8a6f-c34d-ba6b-dfc0-b4e12472e402
* Content: [_HUB_DEVICE_CONFIG_INFO_V1](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/usbioctl/ns-usbioctl-_hub_device_config_info_v1)
* Content Source: [wdk-ddi-src/content/usbioctl/ns-usbioctl-_hub_device_config_info_v1.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/usbioctl/ns-usbioctl-_hub_device_config_info_v1.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-17:
@PageWriter-MSFT this seems to be a dupe of Issue #197. Can you verify and close, if so?

PageWriter-MSFT, 2018-09-13:
The new documentation system uses automation to assemble the reference documentation based on the source files. We are using Visual Studio’s intellisense engine for API discovery. This has the advantage of making our docs consistent with what users see in Visual Studio. Unfortunately, that means that we moved to the actual API names from the more appealing typedef aliases.

Page Writer

closed 2018-09-13 by PageWriter-MSFT

BrianCatlin, 2018-09-13:
Then your automation should be fixed! 
Other structures that reference or incorporate structures refer to them by structure name, not by the tag, so people are going to search for the structure name.

PageWriter-MSFT, 2018-09-13:
About searching: All aliases are added to the SEO list for each topic. Searching for any related term should bring up the correct topic. 

Page Writer

BrianCatlin, 2018-09-13:
That doesn't help people who are browsing the table of contents. Bottom line: From a usability perspective, there is no good reason for using just the tag as over the structure name. If your automation prevents that, then fix your automation! Clearly, you have the structure names if you're adding them to your SEO list, so there is no reason not to use the structure name in the docs. A good argument could be made to list both the structure name and the tag name in the docs