MicrosoftDocs/windows-driver-docs-ddi #197: Wrong structure name

BrianCatlin, 2018-08-16:
Why is this page named for the structure tag (_USB_CYCLE_PORT_PARAMS) instead of the structure type (USB_CYCLE_PORT_PARAMS)? None of the docs or examples ever refer to this structure using the tag.

This mistake is made on many of the USB structure doc pages, but I'm not going to file a bug against each one, because I don't get paid for this

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c3d57229-1c2b-5099-34e6-1bf424678e4a
* Version Independent ID: b940c4d8-9114-3540-aa1c-8ac109125401
* Content: [_USB_CYCLE_PORT_PARAMS](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/usbioctl/ns-usbioctl-_usb_cycle_port_params)
* Content Source: [wdk-ddi-src/content/usbioctl/ns-usbioctl-_usb_cycle_port_params.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/usbioctl/ns-usbioctl-_usb_cycle_port_params.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-17:
@PageWriter-MSFT can you please look into whether @BrianCatlin has found a migration bug?

PageWriter-MSFT, 2018-09-13:
Hi Brian,

The new documentation system uses automation to assemble the reference documentation based on the source files. Since C++ doesn’t have built-in reflection, we are using Visual Studio’s intellisense engine for API discovery. This has the advantage of making our docs consistent with what users see in Visual Studio. Unfortunately, that means that we moved to the actual API names from the more appealing typedef aliases.

Page Writer

closed 2018-09-13 by PageWriter-MSFT

BrianCatlin, 2018-09-13:
The purpose of the samples is to teach people how to do something. The samples refer to the structure name, not the tag name. When someone goes to look up the structure, it cannot be found! How is that good documentation?

PageWriter-MSFT, 2018-09-13:
All aliases are added to the SEO list for each topic. Searching for any related term should bring up the correct topic. If you have an example where the structure cannot be found please let me know. I will address the issue.

Page Writer