MicrosoftDocs/windows-driver-docs-ddi #199: HDR swap chains

GoldenAxe1977, 2018-08-17:
Hi there, is says here the DX surface is always A8R8G8B8 formatted, I was wondering if there was a way to make an HDR10 (R10G10B10A2) swap chain instead. Why is the bit depth of this surface hard-coded? If so, I wonder how Windows 10 can manage to set its desktop to HDR mode, let alone use other swap chain types for fullscreen games.

Thanks!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4c145445-bf4c-a7d0-e2ef-4eeb7790458c
* Version Independent ID: 8703e527-fec5-8568-2651-394708dd9915
* Content: [IDDCX_METADATA](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/iddcx/ns-iddcx-iddcx_metadata)
* Content Source: [wdk-ddi-src/content/iddcx/ns-iddcx-iddcx_metadata.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/iddcx/ns-iddcx-iddcx_metadata.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-17:
Thanks for the question, @GoldenAxe1977 
@mimisasouvanh can you please determine if this is a doc issue or a coding issue and respond appropriately?

GoldenAxe1977, 2018-08-17:
Thanks Eliot. To clarify, I'm using Karl Froelich's IndirectDisplay sample to do some tinkering with virtual displays to add some postprocessing and such before mirroring it back onto a real display, using DesktopDuplication API, and it appears as though there are several places along the pipeline where 8-bit swap chains are the only option. I opened up a question here but it seems Karl's too busy :) So any help you guys could offer would be great.

https://github.com/kfroe/IndirectDisplay/issues/8

EliotSeattle, 2018-08-17:
Related to [Issue #8 in the @Kfroe sample](https://github.com/kfroe/IndirectDisplay/issues/8). @mimisasouvanh maybe bring Karl and Steven in on this if it's something we should doc?

mimisasouvanh, 2018-09-25:
@GoldenAxe1977 I truly apologize for taking so long to respond. I've reached out for an answer and hope to hear back soon.

13thsymphony, 2018-09-26:
@GoldenAxe1977, it's true that Windows supports FP16 and UINT10 swapchains for HDR, but we currently don't support this for indirect displays.

We have samples demonstrating HDR support for wired HDR10 displays: [D3D11](https://github.com/Microsoft/Xbox-ATG-Samples/tree/master/UWPSamples/Graphics/SimpleHDR_UWP), [D2D](https://github.com/Microsoft/Windows-universal-samples/tree/master/Samples/D2DAdvancedColorImages), [D3D12](https://github.com/Microsoft/Xbox-ATG-Samples/tree/master/UWPSamples/Graphics/SimpleHDR_UWP12), among others.

GoldenAxe1977, 2018-09-26:
Thanks for getting back to me. Is there any chance of supporting HDR in the future?

I'm not sure what to do at this point, I'm prototyping displays using the indirect driver scheme and HDR is a pretty basic capability these days.

Or perhaps, can you suggest a work-around? What I'm trying to do is, for example, have a 4K 240hz HDR10 virtual display, which I then intercept and pack those frames into a real 4K 60hz display (the 4K DLPs by Texas Instruments), which then unpack it and play it back at 240hz. 

I need Windows to treat this display as genuine 240hz with HDR10, which AFAIK isn't possible at this point in a real driver. I need an abstraction of the primary display with arbitrary capabilities.

I appreciate this may take some effort from Microsoft's part to fix, it would be really great. 

Also, it would be great if you can increase the default 512 Hz max Windows refresh rate somehow, I haven't been able to create a virtual (SDR) display higher than 480hz yet. Ideally 960hz or more. (2048hz would be perfect actually). I'm not going to assume it's as simple as changing a macro and rebuilding Windows, but if it were, that would be terrific.

mimisasouvanh, 2018-10-31:
Hi @GoldenAxe1977 we have taken note of this ask as a feature request. Thank you!

LionsAd, 2023-01-29:
@mimisasouvanh Was this ever implemented to ensure that IDD can use the FP16 swap chain instead?

mirh, 2023-08-14:
Still no update.. will IddCx 1.10 be the charm?
I suppose `onecoreuap\windows\core\dxkernel\indirectdisplays\classext\cx\swapchain.cpp` is the culprit.