MicrosoftDocs/windows-driver-docs-ddi #205: Field order of _HIDP_BUTTON_CAPS is different from the header file

azyobuzin, 2018-08-19:
In "Syntax" section of "_HIDP_BUTTON_CAPS structure" page, it says that `NotRange.Usage` field follows `NotRange.Reserved1` field. But in hidpi.h in Windows SDK 10.0.17134.0, `NotRange.Reserved1` field follows `NotRange.Usage` field.

Here is `NotRange` field in the present "_HIDP_BUTTON_CAPS structure" page.

```c
    struct {
      USAGE  Reserved1;
      USAGE  Usage;
      USHORT StringIndex;
      USHORT Reserved2;
      USHORT DesignatorIndex;
      USHORT Reserved3;
      USHORT DataIndex;
      USHORT Reserved4;
    } NotRange;
```

And this is `NotRange` field in hidpi.h.

```c
        struct  {
            USAGE    Usage,            Reserved1;
            USHORT   StringIndex,      Reserved2;
            USHORT   DesignatorIndex,  Reserved3;
            USHORT   DataIndex,        Reserved4;
        } NotRange;
```

Likewise, the order is different on "_HIDP_VALUE_CAPS structure" page.

---
#### ドキュメントの詳細

⚠ *このセクションを編集しないでください。 docs.microsoft.com で必須です ➟ GitHub の問題のリンク。*

* ID: 721b4597-12b9-b7e0-63f0-29545e065a8a
* Version Independent ID: 3a3f5f3a-d8ae-0b1e-4cef-d980203fc7d7
* Content: [_HIDP_BUTTON_CAPS](https://docs.microsoft.com/ja-jp/windows-hardware/drivers/ddi/content/hidpi/ns-hidpi-_hidp_button_caps)
* Content Source: [wdk-ddi-src/content/hidpi/ns-hidpi-_hidp_button_caps.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/hidpi/ns-hidpi-_hidp_button_caps.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-20:
Thanks, @azyobuzin for pointing this out.
@PageWriter-MSFT can you please determine if this is a known issues with the scraping tools or if it's a new bug? Thank you!

PageWriter-MSFT, 2018-09-12:
Hi @azyobuzin 

It's a bug in our backend tools. Even though the rendered syntax is different from the header syntax, the datatype representation remains the same. I do agree this is confusing to read. 

Thanks for reporting this issue. I have opened a ticket to get this fixed.

Page Writer

closed 2018-09-12 by PageWriter-MSFT