MicrosoftDocs/windows-driver-docs-ddi #207: Should there be argument types in the API Syntax above?

dachms, 2018-08-23:
void RtlFillMemory(
   Destination,
   Length,
   Fill
);

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b3f78de0-499d-f23f-e8db-5a56955fdd3f
* Version Independent ID: 0fe274b3-a564-27cb-a2b8-18675c2c4255
* Content: [RtlFillMemory macro](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-rtlfillmemory)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-rtlfillmemory.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-rtlfillmemory.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-23:
Thanks, @dachms for the question.
Throwing this to you, @PageWriter-MSFT. There's group of these defined together and all doc'd without arg types, and that make me think the answer is no. But you are my kernel expert. 

These defs are from wdm.h

#define RtlEqualMemory(Destination,Source,Length) (!memcmp((Destination),(Source),(Length)))
#define RtlMoveMemory(Destination,Source,Length) memmove((Destination),(Source),(Length))
#define RtlCopyMemory(Destination,Source,Length) memcpy((Destination),(Source),(Length))
#define RtlFillMemory(Destination,Length,Fill) memset((Destination),(Fill),(Length))
#define RtlZeroMemory(Destination,Length) memset((Destination),0,(Length))

PageWriter-MSFT, 2018-09-12:
Hi @dachms 

Technically these are macros that wrap mem* functions so the syntax cannot show the datatype because it is derived from the header. I do agree that its not easy when you don't see the datatype. I have added the datatype in the parameter description. I hope this improves your experience. These should be live soon.

Thanks for the feedback!

Page Writer

closed 2018-09-12 by PageWriter-MSFT