MicrosoftDocs/windows-driver-docs-ddi #208: Misleading enum member values!

abhinavtyagi, 2018-08-24:
The enum members don't have values in incrementing order but do have the values in increasing order index of bit i.e. BIT(x). The way it is presented on this page is misleading and hence incorrect.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 206b7425-c152-21cb-a087-9c57a5370e8f
* Version Independent ID: 1c013a56-7e94-985a-4098-dffc19890984
* Content: [D3D12DDI_RESOURCE_STATES](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/d3d12umddi/ne-d3d12umddi-d3d12ddi_resource_states)
* Content Source: [wdk-ddi-src/content/d3d12umddi/ne-d3d12umddi-d3d12ddi_resource_states.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/d3d12umddi/ne-d3d12umddi-d3d12ddi_resource_states.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-24:
Thanks, @abhinavtyagi 
@mimisasouvanh can you look into this?

mimisasouvanh, 2018-09-25:
Hello @abhinavtyagi,
I sincerely apologize for taking so long to respond. Are you talking about the Syntax section of the document? 
Thanks!

mimisasouvanh, 2018-09-25:
@abhinavtyagi 
Hi Abhi,
I have inserted a new Syntax with enum member values. It should be published in the next 24 hours or so. 
Thanks!
Mimi

closed 2018-09-25 by mimisasouvanh

abhinavtyagi, 2018-09-25:
I can't correlate the page where I did comment and pointed the problem. If
the comment or page link is included, it'd be possible to go back and see
what have changed.

Thanks,
Abhinav

mimisasouvanh, 2018-10-01:
Here is the link: https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/d3d12umddi/ne-d3d12umddi-d3d12ddi_resource_states