MicrosoftDocs/windows-driver-docs-ddi #211: PCREATE_PROCESS_NOTIFY_ROUTINE_EX parameter CreateInfo note

mlfaw, 2018-08-26:
A note should be added to the `CreateInfo` parameter saying that it will be NULL if the process is exiting.
According to https://msdn.microsoft.com/en-us/library/windows/hardware/ff542860(v=vs.85).aspx

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 86077f3f-9b45-5871-2889-c225c55a54e3
* Version Independent ID: ec88f69c-398c-5af6-a84c-fbd9be3efc47
* Content: [PCREATE_PROCESS_NOTIFY_ROUTINE_EX](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/nc-ntddk-pcreate_process_notify_routine_ex)
* Content Source: [wdk-ddi-src/content/ntddk/nc-ntddk-pcreate_process_notify_routine_ex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/nc-ntddk-pcreate_process_notify_routine_ex.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-27:
Thanks for the feedback, @mlfaw 
@PageWriter-MSFT can you look into this?

PageWriter-MSFT, 2018-09-12:
Hi @mlfaw 

Thanks for pointing out that missing piece of information. I have updated the topic. It should be live soon.

Page Writer

closed 2018-09-12 by PageWriter-MSFT