MicrosoftDocs/windows-driver-docs-ddi #212: RegionSize information error

bad1dea5, 2018-08-27:
> If the MEM_RELEASE flag is set in the FreeType parameter, the variable pointed to by RegionSize must be zero. 

This function will fail with 0xc0000005 you do this. You must give the size of the region, that you used with NtAllocateVirtualMemory.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: dc08af12-0480-fba3-7498-98545e6a62a4
* Version Independent ID: f5771ed8-3447-68a9-e548-0709cfd13587
* Content: [NtFreeVirtualMemory function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-ntfreevirtualmemory)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-ntfreevirtualmemory.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-ntfreevirtualmemory.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-27:
Thank you, @bad1dea5 
I'll look into this.

mikrosk, 2019-04-23:
This is not true, at least not on latest W10 release. Just tried it with *RegionSize = 0 and works all right.

lorihollasch, 2019-04-27:
@bad1dea5 , I'm going to close this and assume that your issue was caused by unclear RegionSize versus *RegionSize language, per @mikrosk. The language should be clear now. Let me know if you are still having an issue.

closed 2019-04-27 by lorihollasch