MicrosoftDocs/windows-driver-docs-ddi #213: Documentation Missing

csm10495, 2018-08-29:
Specifics about the parameters required for this IOCTL (input/output buffer) appear to be missing from the documentation on the website. Also the links to PHYSICAL_ELEMENT_STATUS and PHYSICAL_ELEMENT_STATUS_REQUEST are dead.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 399a040e-533e-373c-0259-22deb062da2d
* Version Independent ID: c36bf3c4-5683-db3d-18fe-2e1042c41234
* Content: [IOCTL_STORAGE_GET_PHYSICAL_ELEMENT_STATUS](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddstor/ni-ntddstor-ioctl_storage_get_physical_element_status)
* Content Source: [wdk-ddi-src/content/ntddstor/ni-ntddstor-ioctl_storage_get_physical_element_status.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddstor/ni-ntddstor-ioctl_storage_get_physical_element_status.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-29:
Thanks, @csm10495 for more feedback!
I know that we're working on a bug to get the prerelease ref topics published. I strongly believe that is why the two links aren't resolving. For the parameter descriptions, I am going to defer to @duncanmacmichael, who is authoring this prerelease page. Unfortunately--or fortunately for him--he is out until 9/10, so please give us a little time.

duncanmacmichael, 2018-09-10:
Hi @csm10495 , thanks for finding this. It looks like this page and a few related ones got missed in a big content migration we did a few months back, so I'll go ahead and move the content over manually to fix things up. It should be live soon, so I'll circle back to let you know when it's available.