MicrosoftDocs/windows-driver-docs-ddi #214: Inconsistency with the WDK headers

SamaelRanger, 2018-08-31:
If you look into \Windows Kits\10\include\10.0.16299.0\km\wdm.h  (or 10.0.17134.0) then you will see 
_IRQL_requires_max_(PASSIVE_LEVEL) for the function (IoBuildSynchronousFsdRequest). But doc says it is allowed to use this function on  "IRQL	&lt;= APC_LEVEL". So where is the error?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: fc00c9a7-e502-e59a-be22-aa87bb3e8286
* Version Independent ID: d45f64ff-8c04-1781-178e-88cca095daaa
* Content: [IoBuildSynchronousFsdRequest function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-iobuildsynchronousfsdrequest#feedback)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-iobuildsynchronousfsdrequest.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-iobuildsynchronousfsdrequest.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-31:
Thanks, @SamaelRanger 
@PageWriter-MSFT can you please look into and rectify the discrepancy?

PageWriter-MSFT, 2018-09-12:
Hi @SamaelRanger 

Thanks for spotting the discrepancy. I have filed a ticket for the feature team to correct this in the header. For now, please read the IRQL value as <=APC_LEVEL.

Page Writer

closed 2018-09-12 by PageWriter-MSFT