MicrosoftDocs/windows-driver-docs-ddi #218: Unclear

nobrassring, 2018-09-04:
"up to the length of the destination buffer" should read something like "following the current contents of the destination string and the destination length is incremented by the number of bytes copied."

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 981946ae-4038-870d-4001-a05dadc47618
* Version Independent ID: 474ba3f6-f6d8-50dd-afcf-90cbbf9fda37
* Content: [RtlAppendUnicodeStringToString function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-rtlappendunicodestringtostring)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-rtlappendunicodestringtostring.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-rtlappendunicodestringtostring.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-09-04:
Tanks, @nobrassring 
@PageWriter-MSFT, please consider this request.

PageWriter-MSFT, 2018-09-12:
Hi @nobrassring 

Thanks for the suggestion. I have updated the Remarks section with the suggested text. They should be live soon.

Page Writer

closed 2018-09-12 by PageWriter-MSFT