MicrosoftDocs/windows-driver-docs-ddi #221: missing clarification about synchronization

pjprajap, 2018-09-05:
what if WdfRequestForwardToIoQueue and WdfIoQueueRetrieveNextRequest gets called from two different threads for same queue object ? 
Adding this details in document will give more clarity to developer about synchronizing manualDispatch Queue.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 16abdc96-df03-59fa-b911-b8603615043b
* Version Independent ID: 0e73ceb4-0071-b278-e3eb-d6f5e0019622
* Content: [WdfIoQueueRetrieveNextRequest function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdfio/nf-wdfio-wdfioqueueretrievenextrequest)
* Content Source: [wdk-ddi-src/content/wdfio/nf-wdfio-wdfioqueueretrievenextrequest.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfio/nf-wdfio-wdfioqueueretrievenextrequest.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-09-05:
Interesting question, @pjprajap. Thanks for asking.
@tedhudek, can you take a look and prioritize this appropriately?

tedhudek, 2018-09-10:
Hi @pjprajap, not sure I follow the question.  If thread 1 is forwarding a request and thread 2 is retrieving it, what is ambiguous?

tedhudek, 2018-09-12:
Closing due to lack of activity, please re-open if needed

closed 2018-09-12 by tedhudek