MicrosoftDocs/windows-driver-docs-ddi #225: Enum is not initialized correctly

nadircazi, 2018-09-06:
From ddk=&gt; D3DWDDM2_0DDI_VIDEO_DECODER_CAP_DOWNSAMPLE           = 0x01

The public version of this above implies that D3DWDDM2_0DDI_VIDEO_DECODER_CAP_DOWNSAMPLE           = 0x0

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4a035656-26c9-aa19-5c64-8ff1703a1e66
* Version Independent ID: e8fe6a4b-37df-0fd1-1976-b53d4ec26db9
* Content: [D3DWDDM2_0DDI_VIDEO_DECODER_CAPS](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/d3d10umddi/ne-d3d10umddi-d3dwddm2_0ddi_video_decoder_caps)
* Content Source: [wdk-ddi-src/content/d3d10umddi/ne-d3d10umddi-d3dwddm2_0ddi_video_decoder_caps.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/d3d10umddi/ne-d3d10umddi-d3dwddm2_0ddi_video_decoder_caps.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-09-06:
Oops. Thanks, @nadircazi for pointing this out. @mimisasouvanh, can you look into this and make sure the doc is correct? Thanks.

nadircazi, 2018-09-17:
A gentle reminder to make this quick change, before I forget this as well.
Thanks!

mimisasouvanh, 2018-09-25:
Hi @nadircazi,
I truly apologize for taking so long to respond. I can't seem to see where the incorrect initialization is. Are you looking at the Syntax section of the doc page?

Thanks,
Mimi

nadircazi, 2018-09-25:
Hi Mimi,

From msdn [page](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/d3d10umddi/ne-d3d10umddi-d3dwddm2_0ddi_video_decoder_caps):
typedef enum D3DWDDM2_0DDI_VIDEO_DECODER_CAPS {
  D3DWDDM2_0DDI_VIDEO_DECODER_CAP_DOWNSAMPLE           ,
  D3DWDDM2_0DDI_VIDEO_DECODER_CAP_NON_REAL_TIME        ,
  D3DWDDM2_0DDI_VIDEO_DECODER_CAP_DOWNSAMPLE_REQUIRED  ,
  D3DWDDM2_0DDI_VIDEO_DECODER_CAP_UNSUPPORTED
} ;
...which implies, D3DWDDM2_0DDI_VIDEO_DECODER_CAP_DOWNSAMPLE = 0x0

BUT, looking at the specs/ddk:
D3DWDDM2_0DDI_VIDEO_DECODER_CAP_DOWNSAMPLE = 0x01

So looks like we need to correct the msdn reference [page](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/d3d10umddi/ne-d3d10umddi-d3dwddm2_0ddi_video_decoder_caps)

mimisasouvanh, 2018-09-25:
Ahh I see now. Thanks for clarifying! Hopefully I can get a fix in by tomorrow. It looks as though many other enumeration pages are affected as well.

mimisasouvanh, 2018-09-25:
Hi @nadircazi 
I've inserted a new syntax. It should be published in the next 24 hours.
Thanks!
Mimi

closed 2018-09-25 by mimisasouvanh

nadircazi, 2018-09-26:
Thanks a lot, Mimi!