MicrosoftDocs/windows-driver-docs-ddi #227: Pfnd3dddiPresent1 2nd parameter

englishb, 2018-09-06:
The second parameter to the Pfnd3dddiPresent1 function is unnamed and has no description or documentation.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: cde12cc4-2c56-1b6d-5dcb-e64a02e5c89b
* Version Independent ID: ef0e8d82-5d04-1e85-9085-7b4f2c533991
* Content: [PFND3DDDI_PRESENT1](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/d3dumddi/nc-d3dumddi-pfnd3dddi_present1)
* Content Source: [wdk-ddi-src/content/d3dumddi/nc-d3dumddi-pfnd3dddi_present1.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/d3dumddi/nc-d3dumddi-pfnd3dddi_present1.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-09-06:
Thanks, @englishb 
@mimisasouvanh can you please get this taken care of? Thanks!

mimisasouvanh, 2018-09-25:
@englishb Hi Ben,
I made the fix. You should be able to see the second parameter in the next day.
Thanks,
Mimi

closed 2018-09-25 by mimisasouvanh