MicrosoftDocs/windows-driver-docs-ddi #231: RtlLongToUInt8 docs not in sync

riverar, 2018-09-08:
One of many examples, `RtlLongToUInt8` appears differently on the web vs. what's in this repository. I was going to correct a bunch of these docs, but they're not in sync with the repo.

Will await guidance.

Web: https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntintsafe/nf-ntintsafe-rtllongtouint8

Repository: https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntintsafe/nf-ntintsafe-rtllongtouint8.md

EliotSeattle, 2018-09-10:
Thanks, @riverar 
@PageWriter-MSFT can you see why this and other rendered pages differ from the source? IS the build system injecting something wrong/right? Thanks.

PageWriter-MSFT, 2018-09-12:
Hi @riverar 

Thanks for pointing that out! There was a bug in our backend tools which was fixed today. So you will find the page fixed with previously missing information.

Apologize for the inconvenience,

Page Writer

closed 2018-09-12 by PageWriter-MSFT