MicrosoftDocs/windows-driver-docs-ddi #240: Incorrect pszDest documentation

shaharlux, 2018-09-13:
Instead of " up to cbMaxAppend bytes " should be " up to cbToAppend bytes "

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 677c95a4-ae90-30e3-9248-f6bbdf42ff87
* Version Independent ID: ea0bee36-5855-5742-43ed-6f876a8602f0
* Content: [RtlStringCbCatNA function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntstrsafe/nf-ntstrsafe-rtlstringcbcatna)
* Content Source: [wdk-ddi-src/content/ntstrsafe/nf-ntstrsafe-rtlstringcbcatna.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntstrsafe/nf-ntstrsafe-rtlstringcbcatna.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-09-13:
Thanks, @shaharlux. This one's yours, @PageWriter-MSFT.

tedhudek, 2019-05-20:
Apologies for the delay @shaharlux this is fixed and will go live later today

closed 2019-05-20 by tedhudek