MicrosoftDocs/windows-driver-docs-ddi #245: What is the purpose of the TcpHeaderOffset field?

russli1993, 2018-09-18:
When TCP/UDP checksum is enabled, can the Miniport driver uses the TcpHeaderOffset field as the number of bytes  from the Dest mac to the start of the TCP/UDP header? This offset is required so that the hardware does not need to parse headers above TCP/UDP

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5f0f8051-e6e2-449a-ecb4-9bafbcef2cb3
* Version Independent ID: ee070510-41ca-8f7e-f265-242fdd3ddacf
* Content: [_NDIS_TCP_IP_CHECKSUM_NET_BUFFER_LIST_INFO](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ndis/ns-ndis-_ndis_tcp_ip_checksum_net_buffer_list_info)
* Content Source: [wdk-ddi-src/content/ndis/ns-ndis-_ndis_tcp_ip_checksum_net_buffer_list_info.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ndis/ns-ndis-_ndis_tcp_ip_checksum_net_buffer_list_info.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

duncanmacmichael, 2018-09-19:
Hi @russli1993 , thanks for the question. Sorry about the missing info - this is actually related to a current rendering bug we have with the docs where any members contained in an unnamed struct or union don't show up on the page. We're working to get that fixed as soon as we can. In the meantime, if you click the "Edit" button in the upper right to see the raw Markdown on the GitHub backend, the member descriptions are there. Here's the text for the Transmit.TcpHeaderOffset member:

"The offset, in bytes, of the TCP header from the beginning of the packet for TCP packets. Miniport drivers can use TcpHeaderOffset to determine the location of the TCP header so that they do not need to parse the MAC and IP header."

So, to answer your question, this offset is from the beginning of the packet to the TCP header, not from the destination MAC address. Hope this helps!

closed 2018-09-19 by duncanmacmichael

russli1993, 2018-09-19:
Thanks. I have additional questions. 
1. Will this always be available? If the hardware is built to always expect this offset and offset is not set by some upper layer driver for some reason, should the miniport driver parse the packet to get to this field?
2. UDP checksum also includes the source IP, destination IP and protocol (the pseudo header), is the pseudo checksum available in the UDP header as well like TCP pseudo checksum? Is the offset to start of UDP header provided by upper layer?

duncanmacmichael, 2018-09-19:
Hi @russli1993 , I don't know those answers but let me ask the feature team if this is something we can clarify for this page.

duncanmacmichael, 2018-09-20:
Hi @russli1993 , I've gotten some answers to your questions. 

1. For TCP packets, yes, the TcpHeaderOffset field is always set by the TCP/IP transport.
2. Yes, the pseudo checksum is available in the UDP header's Checksum field just like it is for TCP headers.
3. No, the offset to the start of the UDP header is not provided by the TCP/IP transport. The TcpHeaderOffset field is not populated by the transport for UDP packets, and unfortunately I don't have an exact reason why that is.

I'll update this page to clarify the TcpHeaderOffset field in the remarks section. I'll also update the [Offloading Checksum Tasks](https://docs.microsoft.com/windows-hardware/drivers/network/offloading-checksum-tasks) page to mention that UDP pseudoheader checksum calculation happens in the same way as for TCP.

Thanks for the questions and helping to clarify the docs!