MicrosoftDocs/windows-driver-docs-ddi #248: Missing kernel routine

Arush-Agarampur, 2018-09-22:
I don't know where else to put this, but the ntddk.h documentation is missing the documentation for the funtion MmCopyVirtualMemory. Is there a way that I can add that info somewhere?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 679192b8-624a-d022-23af-33c76a77c891
* Version Independent ID: f6126a61-c7f7-cac4-97d1-86c582e95117
* Content: [MmCopyMemory function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/nf-ntddk-mmcopymemory)
* Content Source: [wdk-ddi-src/content/ntddk/nf-ntddk-mmcopymemory.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/nf-ntddk-mmcopymemory.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-09-27:
Hi @Arush-Agarampur I don't see **MmCopyVirtualMemory** in ntddk.h.  Can you be more specific about where you found it?

@PageWriter-MSFT fyi

Arush-Agarampur, 2018-09-27:
It seems to be all over the web with a simple google search. Here are some links that use it:

https://ywjheart.wordpress.com/2017/11/20/mmcopyvirtualmemory/

https://github.com/Zer0Mem0ry/KernelReadWriteMemory/blob/master/ReadWrite.c

http://www.rohitab.com/discuss/topic/40524-what-is-writeprocessmemory-kernel-mode-equivalent/ (Scroll down a bit to see it.)

tedhudek, 2019-05-21:
Sorry @Arush-Agarampur but I don't see **MmCopyVirtualMemory** in the headers that shipped with the latest (Windows 10 version 1903) SDK or WDK.

closed 2019-05-21 by tedhudek