MicrosoftDocs/windows-driver-docs-ddi #250: Sr Developer Support Esc. Engineer

Ron-Hu, 2018-09-25:
There is NOTHING defined here.  This page is useless for developers writing miniports.  As long as MSDN exists as a URL.  We should back point readers there if we can't do this equally.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3eb29277-0674-6534-c5b3-33f23c8d1139
* Version Independent ID: 56fc2963-d527-21e9-5411-8d721ce1bc81
* Content: [_HW_INITIALIZATION_DATA](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/storport/ns-storport-_hw_initialization_data~r1)
* Content Source: [wdk-ddi-src/content/storport/ns-storport-_hw_initialization_data~r1.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/storport/ns-storport-_hw_initialization_data~r1.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-09-25:
Dreadfully sorry, @Ron-Hu. Something is happening to this page in which the content in the source is not being picked up by the  build. We are investigating this afternoon. You can see the content if you [click Edit](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/storport/ns-storport-_hw_initialization_data~r1.md). Haven't seen this before. Thanks for bringing it to our attention.

EliotSeattle, 2018-09-25:
Hi, @Ron-Hu.
We have a bug filed with our build team. Apparently, a DDI page with an unnamed struct or union as a member causes the other members to get messed up with the rendering, even though the content is behind the page. I'll update tihs as I hear from them.

EliotSeattle, 2018-11-11:
This has now been fixed. Thanks for your patience.

closed 2018-11-11 by EliotSeattle