MicrosoftDocs/windows-driver-docs-ddi #252: FltGetFileNameInformation: STATUS_FLT_INVALID_NAME_REQUEST and post-IRP_MJ_ACQUIRE_FOR_SECTION_SYNCHRONIZATION

kITerE, 2018-09-27:
FltGetFileNameInformation always returns an STATUS_FLT_INVALID_NAME_REQUEST in a post-operation callback routine for IRP_MJ_ACQUIRE_FOR_SECTION_SYNCHRONIZATION. 

Post-IRP_MJ_ACQUIRE_FOR_SECTION_SYNCHRONIZATION mentioned in description of FltGetFileNameInformationUnsafe, but not here.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 82c69953-3606-33d2-1c71-fa25b72a82da
* Version Independent ID: 67b78c7c-0fdc-ab17-d7d7-53a6b3eaded0
* Content: [FltGetFileNameInformation function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nf-fltkernel-fltgetfilenameinformation)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltgetfilenameinformation.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltgetfilenameinformation.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

rodwiddowson, 2018-09-27:
This is expected.  `Acquire_for`*  causes the FSD to set the IoTopLevelIrp and this is one of the things that causes FltGetFileNameInformation to complain.

Grabbing the name in pre and passing it as the context is your best bet.  Also the release works the other way around.

Finally - this sort of question is often better asked in [here](https://community.osr.com/categories/ntfsd)

rodwiddowson, 2018-09-27:
(ignore me, I was warbling, you knew this - it wasn't a question, it was an observation).  I've deleted my unhelpful comment

I'd assume that CacheFlush does the same, FWIW

kITerE, 2018-09-27:
I just wanted this case (post-IRP_MJ_ACQUIRE_FOR_SECTION_SYNCHRONIZATION) to be described (like [FltGetFileNameInformationUnsafe](https://docs.microsoft.com/windows-hardware/drivers/ddi/content/fltkernel/nf-fltkernel-fltgetfilenameinformationunsafe#remarks))

lorihollasch, 2019-05-21:
@kITerE , thanks for reporting this missing info! I have updated the page to include it.

closed 2019-05-21 by lorihollasch