MicrosoftDocs/windows-driver-docs-ddi #258: Request for RtlSetSaclSecurityDescriptor to be documented?

kyanha, 2018-09-30:
NTDLL exports a routine called RtlSetSaclSecurityDescriptor.  Is there a way that this could be documented?  (I'm working on Dokany, a user-mode filesystem driver that needs to create and populate security descriptors in order to determine the correct inheritance for ACLs using SeAssignSecurity.)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a38fcffb-81ce-004e-8f0b-3f9dbde78e9d
* Version Independent ID: f826a867-4470-b244-2ce6-b6dcb0600cd1
* Content: [RtlSetDaclSecurityDescriptor function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-rtlsetdaclsecuritydescriptor)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-rtlsetdaclsecuritydescriptor.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-rtlsetdaclsecuritydescriptor.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

PeterGV, 2018-10-01:
FWIW, I support this request -- RtlSetSaclSecurityDescriptor has been exported for many years, and -- while it's not super-common to need to be need to build a SACL, it's both a reasonable thing to do.  If people can call RtlSet**Dacl**SecurityDescriptor then they should be able to similarly call RtlSet**Sacl**SecurityDescriptor.

tedhudek, 2019-05-31:
Sorry for the delay @kyanha I've just started looking into this.  Will let you know when I have more.  In the meantime here are related links just for my own reference:

https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-rtlsetdaclsecuritydescriptor

https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-rtlgetsaclsecuritydescriptor

kyanha, 2019-06-03:
Another potentially related link:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-rtlgetdaclsecuritydescriptor

I'm seeing that ntifs.h RtlGetDaclSecurityDescriptor is very close to wdm.h
RtlSetDaclSecurityDescriptor.  ntifs.h RtlGetSaclSecurityDescriptor matches
the prototype of RtlGetDaclSecurityDescriptor (except for variable names).

Can it be assumed that RtlSetSaclSecurityDescriptor matches the prototype
of RtlSetDaclSecurityDescriptor?  i.e.:
```

NTSYSAPI NTSTATUS RtlSetSaclSecurityDescriptor(
  PSECURITY_DESCRIPTOR SecurityDescriptor,
  BOOLEAN              SaclPresent,
  PACL                 Sacl,
  BOOLEAN              SaclDefaulted
);

```

If it does match, then I can probably infer what I need to know (including
the privilege [SE_SECURITY_NAME] and permission [ACCESS_SYSTEM_SECURITY]
check requirements that need to be performed before I pass it over to the
RtlSetSaclSecurityDescriptor routine).  The routine does also need to be
included in either ntifs.h or wdm.h (or some other header) as well, though,
which also involves modification of the DDK package itself.

tedhudek, 2019-06-04:
Hi @kyanha sorry it took a while, but the official decision on this one is not to doc it.  If you need more info for a workaround, I'd start with http://www.osronline.com/showlists.cfm?list=ntdev.

closed 2019-06-04 by tedhudek

PeterGV, 2019-06-04:
Unfortunate outcome, but not exactly a burning issue in my book.

Thanks @tedhudek for driving this to closure.