MicrosoftDocs/windows-driver-docs-ddi #261: arguments to KeReleaseSpinLock

thselvam, 2018-10-01:
is the first argument to KeReleaseSpinLock(), saved IRQL level or pointer to the spinlock?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 69dea4bb-ff9c-8eaf-9146-2eb6e200d702
* Version Independent ID: 871e78c6-04d2-b6c7-932e-7cf3a40e32a3
* Content: [KeReleaseSpinLock macro](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-kereleasespinlock)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-kereleasespinlock.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-kereleasespinlock.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

PeterGV, 2018-10-16:
This is another one of those [badly broken MSDN pages reported in issue 269](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/issues/269), guys.  It shows the function prototype as:

void KeReleaseSpinLock(
   a,
   b
);

Clearly munged somehow.

tedhudek, 2018-10-24:
Hi @PeterGV @thselvam thanks so much for reporting these issues.  We're very sorry about all the problems with WDK DDI docs recently.  It's very much a known issue and we are working to make it right.  One workaround is to clone this repo and search locally.  There's one bug we have where some of the pages in this repo don't get pushed out to docs.ms, so at least you would be able to see those.  Thanks for all the specific instances, this really helps troubleshoot.

rodwiddowson, 2018-10-25:
... Plyus of course if you need to you can look on github.  the linke are all broken and there is a ton of noise at the topo but for instance [FsRtlTeardownPerFileContexts](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-fsrtlteardownperfilecontexts.md)

tedhudek, 2019-05-30:
Just pushed a fix for some of these, waiting for it to prop to the live site

tedhudek, 2019-06-05:
This is fixed now --

https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-kereleasespinlock

https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-keacquirespinlock

closed 2019-06-05 by tedhudek