MicrosoftDocs/windows-driver-docs-ddi #262: Invalid documentation

I-gor-C, 2018-10-04:
The documentation appears to be incorrect. It explicitly links the WRITE_REGISTER_BUFFER_ULONG64 to UMDF. This doesn't seem correct and rest of the WRITE_REGISTER_BUFFER_xxx flavors don't mention it in any way. Moreover, the function signature is incorrect and doesn't match the one in the actual wdm.h header.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: bd7e5c28-cfdd-5eb3-2998-76cd77ab72ee
* Version Independent ID: e6eaafba-f73d-970f-5e17-47b45502bacc
* Content: [WRITE_REGISTER_BUFFER_ULONG64 function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-write_register_buffer_ulong64)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-write_register_buffer_ulong64.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-write_register_buffer_ulong64.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

I-gor-C, 2019-01-16:
Any update on this?

tedhudek, 2019-01-16:
Hey @I-gor-C sorry for the delay on this one.  Good catch.  I fixed https://docs.microsoft.com/windows-hardware/drivers/ddi/content/wdm/nf-wdm-write_register_buffer_ulong64, would you mind verifying that it looks as expected? Thanks again for the feedback

closed 2019-01-16 by tedhudek

I-gor-C, 2019-02-08:
@tedhudek Thanks for handling this!