MicrosoftDocs/windows-driver-docs-ddi #266: Lost (nearly) all the Cc Functions

rodwiddowson, 2018-10-10:
For instance hitting F1 in Visual Studio on `CcCanIWrite` gives a 404.  Equally `CcCanIWrite` (or `CcCopyWrite` and so on) is not in the table of contents.

My (admittedly not up to date) clone does have ```
Directory of D:\Perforce\Github\windows-driver-docs-ddi\wdk-ddi-src\content\ntifs

9/08/2018  09:08             3,787 nf-ntifs-cccaniwrite.md
              1 File(s)          3,787 bytes
```

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0f39a405-8a1a-8d58-a796-05691153a17c
* Version Independent ID: bf989985-3b5f-0304-f20e-bc1a7a2838ae
* Content: [ntifs.h header](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/)
* Content Source: [wdk-ddi-src/content/ntifs/index.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/index.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-10-11:
Thanks, @rodwiddowson.
We have an internal bug open for this issue and the content engineering team is researching cause and solutions. Appreciate your vigilance!

rodwiddowson, 2018-10-23:
`nf-ntifs-fsrtlchangebackingfileobject.md` missing too (but (a) you knew this and (b) I an sort of decode the source.

lorihollasch, 2019-05-31:
@rodwiddowson , just wanted you to know that this issue is not forgotten. Unfortunately I don't have an estimated time for a fix, but we will continue to strive towards one!

lorihollasch, 2019-06-19:
@rodwiddowson , I'm gonna close this issue since we are tracking it with an internal bug. Thanks again for reporting it - apologies for the inconvenience!

closed 2019-06-19 by lorihollasch