MicrosoftDocs/windows-driver-docs-ddi #268: Missing system Attribute Flags

rodwiddowson, 2018-10-16:
Checking against `include\km\ntifs.h` from the WDK version 10.0.0.177763.0 
The following attribute are available but not documented:
```
#define FILE_RETURNS_CLEANUP_RESULT_INFO    0x00000200  // winnt
#define FILE_SUPPORTS_POSIX_UNLINK_RENAME   0x00000400  // winnt
#define FILE_SEQUENTIAL_WRITE_ONCE          0x00100000  // winnt
#define FILE_SUPPORTS_TRANSACTIONS          0x00200000  // winnt
#define FILE_SUPPORTS_HARD_LINKS            0x00400000  // winnt
#define FILE_SUPPORTS_EXTENDED_ATTRIBUTES   0x00800000  // winnt
#define FILE_SUPPORTS_OPEN_BY_FILE_ID       0x01000000  // winnt
#define FILE_SUPPORTS_USN_JOURNAL           0x02000000  // winnt
#define FILE_SUPPORTS_INTEGRITY_STREAMS     0x04000000  // winnt
#define FILE_SUPPORTS_BLOCK_REFCOUNTING     0x08000000  // winnt
#define FILE_SUPPORTS_SPARSE_VDL            0x10000000  // winnt
#define FILE_DAX_VOLUME                     0x20000000  // winnt
#define FILE_SUPPORTS_GHOSTING              0x40000000  // winnt
```

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c1946dad-51f1-6553-d6ac-ceba66a3a733
* Version Independent ID: 3e40a24e-7072-d101-980d-0b8f034e3c84
* Content: [_FILE_FS_ATTRIBUTE_INFORMATION](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/ns-ntifs-_file_fs_attribute_information)
* Content Source: [wdk-ddi-src/content/ntifs/ns-ntifs-_file_fs_attribute_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/ns-ntifs-_file_fs_attribute_information.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

PeterGV, 2018-10-16:
Just to add a +1 to this... In case it's not intuitively obvious, these types of attributes are uber-important for Minifilters.  I'm particularly interested in a few of these, such as "SUPPORTS_TRANSACTIONS"

Getting these documented would be useful.

lorihollasch, 2019-06-07:
@rodwiddowson , thanks for reporting the missing flags. I've added them along with descriptions. I've filed an internal task for myself to evaluate what additional information might need to be added to the conceptual docset around these flags. If you have any thoughts around this, feel free to email me! :)

SullyPanda76cl, 2023-04-16:
The page at Microsoft Learn that documents this atributes has a bug:
[fileapi-getvolumeinformationW](https://learn.microsoft.com/en-us/windows/win32/api/fileapi/nf-fileapi-getvolumeinformationW)
[fileapi-getvolumeinformationA](https://learn.microsoft.com/en-us/windows/win32/api/fileapi/nf-fileapi-getvolumeinformationA)

Says:
`FILE_SUPPORTS_GHOSTING = 0x00000200`

But the correct value is
`FILE_SUPPORTS_GHOSTING = 0x40000000`

closed 2023-04-20 by lorihollasch