MicrosoftDocs/windows-driver-docs-ddi #269: Many WDK Doc Pages Missing or Hideously Broken

PeterGV, 2018-10-16:
Something serious is going on with the MSDN WDK doc pages... and it's not good.

For one thing, several pages related to the Cache Manager are just plain missing.  For example, CcCopyWrite… where did it go?

Or... look at the page for [IOCTL_DISK_GET_CACHE_INFORMATION](https://docs.microsoft.com/en-us/windows/desktop/api/winioctl/ni-winioctl-ioctl_disk_get_cache_information) -- Several parameters are missing from the call to DeviceIoControl.

There are many such isolated issues.  I was told by an associate that this was a "known issue" so I put off reporting it until now.  But, more than a week later, I see the issues persist.  Soooooo… I'm reporting this here.

Peter

rodwiddowson, 2018-10-16:
Yea, when I saw #266 closed I assumed that you just had to revert the tolling to yesterday, or failing that just revert the webpages to yesterday.  But that was last week and it’s beginning to be a drag

tedhudek, 2018-11-01:
To update on this issue @PeterGV @rodwiddowson we've removed the redirects for 1258 files that exist in this repo but are for complex reasons not appearing on docs.ms.com. This means that the original MSDN page for each one is again accessible. Once search engines catch up, this should get you what you need in the short term, although we realize the cross-site cognitive dissonance is not the best user experience. Thanks for your patience and please bear with us as we get the new system ironed out.

closed 2018-11-01 by tedhudek

rodwiddowson, 2018-11-02:
Thanks @tedhudek