MicrosoftDocs/windows-driver-docs-ddi #271: ExInterlockedPopEntrySList returning VOID

russli1993, 2018-10-19:
Does ExInterlockedPopEntrySList  return VOID or return the popped SLIST_ENTRY?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 44edf237-f785-1e33-dc4a-6894db868fca
* Version Independent ID: 2efdd705-fb0f-12d0-ff2a-1df1c63c1dcd
* Content: [ExInterlockedPopEntrySList macro](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-exinterlockedpopentryslist)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-exinterlockedpopentryslist.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-exinterlockedpopentryslist.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

sirotnikov, 2018-10-31:
Seems to return the popped PSLIST_ENTRY, as per wdm.h declarations

tedhudek, 2019-05-31:
Hi @russli1993 you are correct! Thanks for the feedback and apologies for the delay.  I've just pushed a change to fix this.  It should appear on the live site within a half hour or so.  Thanks @sirotnikov for the confirmation

closed 2019-05-31 by tedhudek