MicrosoftDocs/windows-driver-docs-ddi #275: Documentation ambiguous on periodic timers

jakeoshins, 2018-10-22:
This documentation doesn't make it clear whether the driver should call WdfTimerStart if provides a non-zero Period in the timer config struct.  Does a periodic timer start ticking as soon as it is created?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: dfb50c08-425e-3cf7-7723-0ea1651e0e83
* Version Independent ID: 8a06b0e1-faf0-254a-4a09-da3318a8be9d
* Content: [WdfTimerCreate function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdftimer/nf-wdftimer-wdftimercreate)
* Content Source: [wdk-ddi-src/content/wdftimer/nf-wdftimer-wdftimercreate.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdftimer/nf-wdftimer-wdftimercreate.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

duncanmacmichael, 2018-10-23:
Hi @jakeoshins, thanks for the question. You're right, the doc isn't explicit about this currently. The closest it gets is this: "After creating a timer object, the driver calls WdfTimerStart to start the timer's clock." This seems to imply the answer to your question is no, a periodic timer does not automatically start ticking. Anecdotally, I had a periodic timer in a driver that I wrote and I did call WdfTimerStart() after initializing it.

But, let me find out the definitive answer for you from someone on the WDF team.

duncanmacmichael, 2018-10-24:
OK @jakeoshins, I got an answer from the WDF team that you still have to call WdfTimerStart() regardless of whether the timer is periodic or not. I've added this wording to this page, [WdfTimerStart](https://docs.microsoft.com/windows-hardware/drivers/ddi/content/wdftimer/nf-wdftimer-wdftimerstart), and [Using Timers](https://docs.microsoft.com/windows-hardware/drivers/wdf/using-timers). Those changes should be live later today. Thanks for helping to improve the docs!

closed 2018-10-24 by duncanmacmichael