MicrosoftDocs/windows-driver-docs-ddi #276: NtQueryInformationFile call

abhay80, 2018-10-23:
Documentation says that we need to pass the handle created by a successful call to NtCreateFile or NtOpenFile. Can we also pass file handle returned from win32 api CreateFileW API call to NtQueryInformationFile()?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5664a237-7ea2-0f4a-2197-99cb9637335e
* Version Independent ID: 027d3a75-ab0b-aac0-61e4-d27bb5531eec
* Content: [NtQueryInformationFile function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-ntqueryinformationfile#feedback)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-ntqueryinformationfile.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-ntqueryinformationfile.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

rodwiddowson, 2018-10-23:
You don't specify the mode you are calling in - this doesn't change answer, but it sound't like you are wanting to call in user mode and you need to know that this documentation is targeted at kernel mode developers.

the answer is yes.  

And in kernel mode you van add `ZwCreateFile`, `FltCreateFileXX`, `IoCreateFileYY` and even `ObOpenObjectByPointer` (if you know what you are doing).   Indeed calling `NtCreateFile` is a minority API for kernel mode.

To hijack this into a Documentation suggestion: It feels to me that the opening paragraphs need to clarify the usermode/kernel mode thing, and then for the kernel mode list the APIs I mention above, do not mention `NtXXX`, but rather have a reference to the very excellent [Using Nt and Zw Versions of the Native System Services Routines|Using Nt and Zw Versions of the Native System Services Routines].  Then usermode could mention `CreateFileZZ` (for Win32 subsystems only) as well as the native API.  I know it crosses architectural boundaries; I suppose if you wanted to avoid Win32/Native API pollution, you might want to reference people to that weird new WIN32 API which is something like `QueryInformationByHandle` 

FWIW

abhay80, 2018-10-23:
Thank you for your quick feedback. 
My apologies as I was not clear about mode of operation. To be clear my question was more specific to user mode as my application works in the user mode.

abhay80, 2018-10-23:
I have worked in user and kernel mode both, so can understand about Zwxxx  and Ntxxx APIs usages. I didn't find any user mode documentation for NtQueryInformationFile function so referred to this kernel mode one. As you have mentioned there should be some clarity about using file handle from createfilezz and NT api version call can be passed to NtQueryInformationFile when calling from user mode. Is it possible to modify the documentation accordingly? 
My query was more specific to NtQueryInformationFile because, IMO, checking for case sensitivity of folder can only be done through this API. I am not sure if win32 version of Apis QueryInformationByHandle or QueryInformationByHandleEx has the capability to provide case sensitivity information of the folder. Could you please confirm?

lorihollasch, 2019-04-27:
@abhay80 , looks like you got the guidance you needed (thanks @rodwiddowson ). I did add "or equivalent" language to NtQueryInformationFile (there are several possibilities, too numerous to call out).

closed 2019-04-27 by lorihollasch