MicrosoftDocs/windows-driver-docs-ddi #283: API should return something for caller to know result.

pjprajap, 2018-10-29:
ddk header file says, API has return value of type BOOLEAN, which is missing here.  Otherwise, how caller suppose to know result where bit is set or not ? 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6260b0d3-985d-5996-37cc-c6bd6d56df1a
* Version Independent ID: fc393953-7eb7-d406-8de2-0fdf77a0781c
* Content: [RtlCheckBit macro - Windows driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-rtlcheckbit)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-rtlcheckbit.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-rtlcheckbit.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2019-05-30:
Fixed, waiting for live site to pick it up

tedhudek, 2019-06-05:
Fixed at https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-rtlcheckbit.  Thanks @pjprajap for the find

closed 2019-06-05 by tedhudek