MicrosoftDocs/windows-driver-docs-ddi #295: SpinLockMem should mention NonPagedPool

Ron-Hu, 2018-11-07:
We should remind the reader to make sure they are providing the spin lock from non-paged pool as you can't Page fault at DPC

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 92b11983-8e0d-cc24-2aa3-f0f687bab035
* Version Independent ID: ca93b4c7-917f-74b8-90de-706469f3c477
* Content: [KeAcquireSpinLockAtDpcLevel macro - Windows driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-keacquirespinlockatdpclevel#feedback)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-keacquirespinlockatdpclevel.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-keacquirespinlockatdpclevel.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2019-05-30:
Thanks for the feedback @Ron-Hu and apologies for the delay.  This is fixed and will appear on the live site within the hour.

closed 2019-05-30 by tedhudek

Ron-Hu, 2019-07-12:
Great! (y)