MicrosoftDocs/windows-driver-docs-ddi #297: This should include example output.

mstaylorh, 2018-11-09:
"{xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx}" for example, to make it clear that {} are included in the returned string.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4ffb3f47-df32-3b09-fcb9-66d5b67fdcf5
* Version Independent ID: 1198defe-3679-4e75-5728-67f90bf32790
* Content: [RtlStringFromGUID function - Windows driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-rtlstringfromguid)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-rtlstringfromguid.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-rtlstringfromguid.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2019-05-31:
Hi @mstaylorh thanks for the feedback and apologies for the delay.  I've just pushed a change to fix this.  It should appear on the live site within a half hour or so.

closed 2019-05-31 by tedhudek

mstaylorh, 2019-05-31:
Looks good, thanks!