MicrosoftDocs/windows-driver-docs-ddi #300: Links to GUID_DEVICE_RESET_INTERFACE_STANDARD are bad

peterwie, 2018-11-13:
This page contains several dead links to https://msdn.microsoft.com/library/windows/hardware/dn928420

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c12fb3d8-6b19-d0ca-08f7-32e7a481fd71
* Version Independent ID: ef9f2e32-8d07-43eb-c498-a2eba9eee103
* Content: [PDEVICE_RESET_HANDLER - Windows driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nc-wdm-pdevice_reset_handler)
* Content Source: [wdk-ddi-src/content/wdm/nc-wdm-pdevice_reset_handler.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nc-wdm-pdevice_reset_handler.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

DOMARS, 2018-11-15:
@peterwie - It looks like the GUID_DEVICE_RESET_INTERFACE_STANDARD topic was missed in the shift to docs.microsoft.com. I will work to add it back and re-establish the links.

DOMARS, 2018-11-16:
@peterwie - I recreated the GUID_DEVICE_RESET_INTERFACE_STANDARD topic and fixed the links from four topics that had linked to it. Thanks for calling this out.

closed 2018-11-16 by DOMARS