MicrosoftDocs/windows-driver-docs-ddi #301: "Which" parameter description is wrong.

ladislav-zezula, 2018-11-13:
The DEBUG_OUTCBI_XXX flags are intended for the interest mask (IDebugOutputCallbacks2::GetInterestMask). For this function, DEBUG_OUTCB_XXX constants must be used, which have different values.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 176ec38d-fb30-acfd-2a1d-7b69e470dd58
* Version Independent ID: 7b03e5e6-b114-b7b4-f3c0-e09e01ed52c8
* Content: [IDebugOutputCallbacks2::Output2 - Windows driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgeng/nf-dbgeng-idebugoutputcallbacks2-output2)
* Content Source: [wdk-ddi-src/content/dbgeng/nf-dbgeng-idebugoutputcallbacks2-output2.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgeng/nf-dbgeng-idebugoutputcallbacks2-output2.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

DOMARS, 2018-11-13:
@ladislav-zezula - Yes it looks like the DEBUG_OUTCB_XXX was confused with  DEBUG_OUTCBI_XXX. Fixes are on the way and should be published in a day or so. Thanks for raising this issue.

DOMARS, 2018-11-29:
@ladislav-zezula - Yes, it looks like the CB was missed. Fixes are on the way. Thanks for your patience.

DOMARS, 2018-11-30:
@ladislav-zezula - The which description now reads:

The kind of DEBUG_OUTCB_XXX notification that is coming in. The DEBUG_OUTCB_XXX notifications are defined in the dbgeng.h header using #defines. For more information, see DEBUG_OUTCB_XXX.

So I will close this issue. Thanks for validating the fix.

closed 2018-11-30 by DOMARS