MicrosoftDocs/windows-driver-docs-ddi #303: Document cases in SystemManagedIdleTimeoutWithHint where "IdleTimeout" is advisory

tzimmer1, 2018-11-16:
We have some teams at my company that refuse to use SystemManagedIdleTimeout because "IdleTimeout" is documented here as advisory and teams are concerned about responsiveness (i.e. taking a Dx IRP too fast).

It would be helpful to have a brief explanation of cases where "IdleTimeout" would be advisory.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a2f0deca-af1b-639f-81d9-b1238af8c3c5
* Version Independent ID: 9e6d98c6-aeee-e5a7-7700-e5a3abcdb852
* Content: [_WDF_POWER_POLICY_IDLE_TIMEOUT_TYPE - Windows driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdfdevice/ne-wdfdevice-_wdf_power_policy_idle_timeout_type)
* Content Source: [wdk-ddi-src/content/wdfdevice/ne-wdfdevice-_wdf_power_policy_idle_timeout_type.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfdevice/ne-wdfdevice-_wdf_power_policy_idle_timeout_type.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2019-06-03:
Hi @tzimmer1 thanks for the question and apologies for the delay.  I'm investigating internally and will let you know what I find.

tedhudek, 2019-06-05:
Thanks again for the feedback @tzimmer1.  I just updated the doc page (should be live in ~30 min).

The updated description for **SystemManagedIdleTimeoutWithHint** now reads:

Starting in Windows 8, the PoFx uses the value specified in the <b>IdleTimeout</b> member as an input when determining at what point after all the components are idle to transition the device to a low-power (Dx) state. This option enables the driver to delay the transition to a low-power state.

The actual duration after which the PoFx allows the device to enter a low-power state might be greater than the <b>IdleTimeout</b> value.

On operating systems earlier than Windows 8, the behavior is the same as <b>DriverManagedIdleTimeout</b>.

closed 2019-06-05 by tedhudek