MicrosoftDocs/windows-driver-docs-ddi #313: Classification field needs more explanation

BrianCatlin, 2018-11-30:
It should be noted that the Classification field is set by the ELAM driver. Why aren't the legal values listed?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4080585d-6e46-b6c7-115d-dfa97477eb33
* Version Independent ID: 181e5da7-4032-e667-31af-14460bf72d4a
* Content: [_BDCB_IMAGE_INFORMATION - Windows driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/ns-ntddk-_bdcb_image_information)
* Content Source: [wdk-ddi-src/content/ntddk/ns-ntddk-_bdcb_image_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/ns-ntddk-_bdcb_image_information.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2019-06-12:
Hey Brian, I've been unable to confirm that the ELAM driver sets the Classification field of the BDCB_IMAGE_INFORMATION structure.  But I did add a link to the BDCB_CLASSIFICATION enum in the field description for Classification on https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/ns-ntddk-_bdcb_image_information.  Thanks again for the feedback.

closed 2019-06-12 by tedhudek

BrianCatlin, 2019-06-12:
If you look at the documentation for BDCB_CLASSIFICATION, it is very clear that the ELAM driver must set this field. How couldn't the devs confirm this?

tedhudek, 2019-06-12:
Not being an ELAM expert, what I got from the docs was that the ELAM driver can call IoRegisterBootDriverCallback to register a BOOT_DRIVER_CALLBACK_FUNCTION, which Windows
then calls, providing a ptr to a BDCB_IMAGE_INFORMATION struct, which has the Classification member already set.  Maybe you can clarify if I'm mistaken?

BrianCatlin, 2019-06-12:
No, the ELAM driver callback is supposed to scan the executable and then report its findings in the BDCB_IMAGE_INFORMATION.Classification variable, specifically whether the executable is good (BdCbClassificationKnownGoodImage) or bad (BdCbClassificationBadImage or BdCbClassificationBadImageBootCritical), which the system uses to determine whether the executable is allowed to run. Setting the variable is the whole point of an ELAM driver

tedhudek, 2019-06-18:
Thanks Brian, I just pushed a change to the description of the Classification member to indicate that it is set by the ELAM driver.